MEGA65/mega65-core #274: compile option USE_LOCAL_<tool>, CC65, ACME, GHDL, etc

Ben-401, 2020-08-11:
As discussed in #220 and etc, We want to be able to tell the MAKE-process to either:
- automatic download/compile/use the required tools (to allow a simple build process), or
- use our locally installed versions of specific tools (to reduce compile time etc.)

Previous attempts to implement this feature have had partial success.

This issue continues-on with this feature request, and also widens the scope to support other tools.
Other tools may include: cc65, acme, ghdl, ophis, iverilog, etc...

USAGE:
The default is to download/compile/use all required tools.
To use your locally installed version:
- you need to have the tool-executable in the $PATH, and
- use the "USE_LOCAL_TOOL=1" in the make command,  
ie: ```make USE_LOCAL_ACME=1 [target]```

Ben-401, 2020-08-11:
i now plan on implementing the same logic to the USE_LOCAL_GHDL

Ben-401, 2020-08-11:
this PR is ready for check/merging.

lydon42, 2021-12-21:
Merged in master/development/hwsc.
Closing.

closed 2021-12-21 by lydon42