MEGA65/mega65-core #277: Add mechanism to reliably detect MEGA65 models

gardners, 2020-08-13:
The I2C initialisation for the MEGA65 varies based on the target, e.g., M65R3 vs earlier revisions vs N4DDR boards etc.

This is important, because otherwise we would need complex detection logic in HYPPO which would use up precious space in the little HYPPO memory.

Ben-401, 2020-08-20:
Hi Paul,
How would a user verify that this feature is working as intended?
Is it as easy as "PRINT( PEEK$D629 )"?

gardners, 2020-08-20:
It differs slightly between C64 and C65 mode.  In C64 mode, you can set the M65 IO mode easily:
POKE 53295,ASC("G"):POKE 53295,ASC("S"):PRINT PEEK(54825)

In C65 mode, its a bit trickier, because you can't use POKE to switch IO mode to MEGA65 mode.  But once it is enabled, you can use PEEK(HEX"D629") or PEEK(54825) after that.

Paul

gardners, 2020-09-11:
This function is working and used in some software already.

closed 2022-01-09 by lydon42