MEGA65/mega65-core #279: Shift is stick for Cursor-Up and Cursor-Left

deftmega, 2020-08-16:
When moving cursor up or left and immediately pressing for example "S" key, the shifted key (in case of "S" the "<3 (heart)" will appear instead. Shift is behaving as if it is released to slowly.

Ben-401, 2020-08-20:
Hi Deft,
What target hardware are you testing on?
And what branch/commit are you using?

And is this behaviour seen directly from a cold start, or do you need to do-something before this issue appears like accessing the sdcard or moving joystick?

deftmega, 2020-08-20:
Hi Ben,

it happens in C64 Basic right from the start on my R3 board with mega65r3-20200817-15-138-hd-cbf5b89+DIRTY and the 5 before that in combination with CPLD firmware mega65-kbd-cpld-222-d3c3ee7a.tgz

cheers
deft

closed 2021-12-20 by deftmega