MEGA65/mega65-core #283: Needed workaround for IEC transfer in 40 MHz mode

FeralChild64, 2020-09-03:
In 40 MHz mode probably every single disk fast-loader will fail, also the C64 ROM routines are most likely incompatible.

Both Turbo Chameleon and Ultimate 64 implemented a workaround: after the software changes something on the IEC bus, the CPU returns to 1 MHz speed for the next 1000 cycles. Something similar is needed for MEGA65.

gardners, 2020-09-11:
Ok, I have implemented 1ms of disabling POKE0,65 (and effect of holding capslock) whenever the IEC port changes state (by drive or by computer).
C65 3.5MHz mode and explicitly selecting 40MHz in MEGA65-specific software will still work, however.
Not yet tested.

FeralChild64, 2020-09-12:
Great! Open ROMs switches speed to 40 MHz by setting $00, so it should be compatible.

lydon42, 2024-08-17:
This is handled by #736 now, so we can close it.

gardners, 2024-08-17:
I think this can indeed be closed @lydon42   , as it has been implmented, and for new software #766 will make the problem obsolete.

closed 2024-08-17 by lydon42