MEGA65/mega65-core #284: AUDIO processing : Add a BBD effect section in the audio processing

jmorel33, 2020-09-12:
This was raised on September 10th 2020 in the MEGA65 youtube channel in the comments aof a video ( https://www.youtube.com/watch?v=NJbSPwVOPhE ), that it would be interesting for developers and sound designers to have the option to use a Reverb / Delay / Chorus for in game effects.  This thread got some traction with many people liking the proposal.

The thought was to include in the audio mixer, a possibility to route the audio of one or multiple channels into an effects chain, or effects module (DSP type processing) to allow a digital form of BBD (Bucket Brigade Delay) type of sound processing.

We asked what would be the basic levers that would be necessary in order to satisfy some of the common requirements of a BBD.  First we need to understand how a BBD works.  For this we could refer to the following thread on Wikipedia.  https://en.wikipedia.org/wiki/Bucket-brigade_device

What was suggested was to allow the following dials to be accessible (via peek/poke memory addresses)
- Delay time (length)
- Delay feedback decay time
- LFO pitch depth (this is a free running LFO)
- LFO rate
- Low Pass filter frequency (this affects the feedback loop) (more fancy would be to have low pass, band pass and high pass with Q factor)

We did not mention if there would be a need for a stereo  model or multiple instances of a BBD, but to at least have one instance would greatly enhance the capabilities vs having no effects at all.  This is why I would suggest to dig deep into this possibility as it may not be out of reach to simply add a sampling algorithm and feedback loop with a filter, and then maybe on a stage two,, add an LFO on the pitch.  The end results will a variety of Reverbs, Delays and Chorus effects, and even possibly phasing.  All of which are the basic effects you can find in the professional music domain for all to enjoy.

NOTE ON MYSELF:  I am a huge fan of the SID chip and the commodore 64.  Also a huge fan of 8-bit and 16-bit computer as a whole, and have been since my first home computer way back in 1985.  I am also a musician (m4ntr0n1c).  You can view my EQUIPBOARD here https://equipboard.com/m4ntr0n1c

gardners, 2020-09-12:
I'm quite happy to consider this, but someone will need to supply a VHDL BBD model that we can look to incorporate. Ideally it would not use any BRAM, as on A100T targets we have none left. Using some distributed RAM is fine, as is using a reasonable number of DSP48 blocks.

jmorel33, 2020-09-13:
So far this is what I found, whcih is a model of the most common BBD chip;
https://www.electrosmash.com/mn3007-bucket-brigade-devices

jmorel33, 2020-09-17:
I haven't found a Bucket Brigade source code... BUT.... Everything else, ,and it is source code which comes from a book, but all available on github https://github.com/juandagilc/Audio-Effects