MEGA65/mega65-core #285: HYPPO: please consider adding some query interface (maybe using some trap?)

lgblgblgb, 2020-09-12:
In some cases it would be useful if a user program (let it be some "system level" stuff - like a test utility etc - more, than an average - let's say - game, etc) to be able to query some information Hyppo has already, just it's not easy to access (as far as I know, correct me, if I'm wrong). What I can mention:

* current bitstream version as string (or later, as kind of numeric value **too**?). _As hickup displays version string on reset, it surely knows it but AFAIK it's not possible easily a userpace program to "extract" it_
* SD-card detected max sector number (per SD bus?). _For some programs, want to do direct SD I/O (yes, even if it's not so much advertised to use in the future, some time it is still useful at least after some trap which allows in the future to turn on userspace usability of SD I/O at least)_

I'm not sure on the second how useful it is, but surely, it would be handy to do, as FDISK would not need to do its own.

**What we have already (AFAIK):**

* hardware information: by reading port $D629
* HDOS version number (using a trap)

gardners, 2020-09-13:
Raw SD access will only be allowed after asking hyppo for permission.  Not yet enforced. Then it is up to programmes that do this to size the card themselves, as else we use up precious hyppo code space.

gardners, 2020-09-13:
The bitstream identifier would, however, be good to make available, even if the whole idea is that eventually it won't be necessary.

lgblgblgb, 2020-09-13:
I'm thinking that even currently available information (like bitstream version string, etc etc) should be kept within a single block of memory inside of HYPPO. And just adding a simple trap function which copies that block to "user memory" on request. However for sure the structure of that range should be kept the same (maybe appending something if needed but then caller program must aware which version the "info block" has, and how much space it needs). I think, this way, the memory usage in hypervisor memory is minimal, as there are no separate functions of query things etc, just a simple single "copy the info block for me, as-is" kind of call. That wouldn't require more than "few bytes" (ok, this is a bit too much to say, but anyway ...) to implement. The rest of the decoding logic of that information is deferred to the user program want to query this kind of information.

gardners, 2020-09-13:
Bitstream ID can just be in registers. It's easier and safer that way, and doesn't need a hyper-trap to read it that way.  We already have that for the keyboard CPLD version.

lgblgblgb, 2020-09-13:
It was just an example for a broader term of all available information hyppo knows either by "hardwired" (eg textual name of the firmware with git short commit id, or what it is ...) or detected things maybe on power-on. Honestly my idea came to my mind that I'm writing a "mega65 information" utility and I would also like to display that same text for the "firmware" (ie: bitstream) what hyppo also displays on reset. Surely, it's only a marginal need, but there can be so much information still, besides this "nuance". But if it sounds like a stupid idea, just forget it :)

gardners, 2020-09-15:
Itsa a nice idea to have this. You can also query the flash to get the stored information about the bitstream, and there are FPGA registers available that will tell you which slot you have loaded the bitstream from.
The commit ID from HYPPO is not trustable, though, as it comes from HYPPO, which can be changed via HICKUP.M65.

For the keyboard CPLD, you can query:

GS $D62A KBD:FWDATEL LSB of keyboard firmware date stamp (days since 1 Jan 2020)
GS $D62B KBD:FWDATEH MSB of keyboard firmware date stamp (days since 1 Jan 2020)
GS $D62C KBD:FWGIT0 LSB of keyboard firmware git commit
GS $D62D KBD:FWGIT0 2nd byte of keyboard firmware git commit
GS $D62E KBD:FWGIT0 3rd byte of keyboard firmware git commit
GS $D62F KBD:FWGIT0 MSB of keyboard firmware git commit

Ideally I would provide some similar registers for the main FPGA.

lgblgblgb, 2020-09-17:
Ok, thanks! And btw also HYPPO version is nice to know, what is trustable then at least if no other info it is from there :)

lydon42, 2023-03-30:
This seems like an ancient issue, closing.
Please create new issue with reference to this if still something should be added.

closed 2023-03-30 by lydon42