MEGA65/mega65-core #287: Documentation for VIC-IV XPOS ($D050/$D051) is wrong

JesperGravgaard, 2020-09-13:
In the documentation it appears that the XPOS is available in bits 0-6 of $D050 and all 8 bits of $D051.

However the XPOS value is actually available in 
- $D050 Read horizontal raster scan position (lower byte)
- $D051.0-5 Read horizontal raster scan position (upper 6 bits)

Source  https://github.com/MEGA65/mega65-core/blob/master/src/vhdl/viciv.vhdl

Documentation:
```
        elsif register_number=80 then
          -- @IO:GS $D050.0-5 VIC-IV:XPOS Read horizontal raster scan position
          null;
        elsif register_number=81 then
          -- @IO:GS $D051 VIC-IV:XPOS Read horizontal raster scan position
          null;
```

Actual definition:
```
        elsif register_number=80 then
          fastio_rdata <= std_logic_vector(xcounter_drive(7 downto 0));
        elsif register_number=81 then
          fastio_rdata(7 downto 6) <= "00";
          fastio_rdata(5 downto 0) <= std_logic_vector(xcounter_drive(13 downto 8));

```

gardners, 2020-09-14:
Hi Jesper,

Thanks for picking this one up! I have updated the documentation lines now.

Paul.

dansanderson, 2024-07-01:
This issue can be closed.

closed 2024-07-02 by lydon42