MEGA65/mega65-core #291: review gitstring generated by version.sh

Ben-401, 2020-09-21:
Currently the ./src/version.sh generates a string (which is appended to ```GIT: ```)
```
138-hd,295c309+DIRTY,20200921.20
```

It is suggested that this string be revised.
Does anyone have any ideas on how this can be better?

We are restricted to 37 chars, i think.

Ben-401, 2020-09-21:
currently the format is:
5 chars: "GIT: " +
6 chars, branch info, +
6 chars, repo status lus dirty bit, +
11 chars, date stamp.

Ben-401, 2020-09-21:
i suggest the space be removed from the "GIT: "

for the branch info, i think first three letters of branch (to help with current branch names), dot dot, and last three letters of branch name, if the branchname is 8 chars or less, padd with leading space.

for dirty state, maybe just an asterix or tilda.

for the date, if we are pressed for room, can we remove the 2020, or just make it 20?

Ben-401, 2020-09-21:
can we also have the targetname? ie nexys4ddr-widget or mega65r3
as this will help with screenshots of hhypervisor booting,.
up for discussion.

gardners, 2020-09-21:
hmmm... keeping git helps people know what it is. But get rid of either the : or the space after the :.
Get rid of +dirty replacing it with only ~ so that we still have an indication of this, until we fix up the gitignore etc to ignore changed files in the submodules, or generally figure out a way to solve that problem.
In terms of having the target in there, I'm not sure that that helps much, as the target is visible from the hardware you are running.  At most the target should just be the two-digit target ID we put int $D629, I think, so that we don't waste lots of space on it.  Branch name and commit are more useful.  Also commit number since branch would also be helpful, like jenkins uses, as relating back to jenkins is useful.

Ben-401, 2020-09-26:
Yes OK, targetname not required in longname format, maybe the two-digit code would suffice, 

Id prefer to see "branchname, date, then commit" (opposed to date at the RHS) , so that the newest file  on sdcard/outputdir is at the top.

From memory, the TILDA shows up as the RIGHT-ARROW (based on charset of hypervisor), suggesting dirty.

Regarding the incremental number (like jenkins uses), that number is specific to the Jenkins Project, and not available to local builds. Can I suggest an incremental number that will increment with every commit, gained using the following:
```
git log upstream/ghdl-crash-20200604..138-hdmi-audio-27mhz --oneline | wc -l
493
```
so currently 493 commits since the common commit where "ghdl-crash-20200604" branched off.

Alternatively we could place a "tag" anywhere we like and get a count from there...
I prefer the former.

Ben-401, 2020-09-26:
further, actually Id want the branchname to be more than 6 chars, we have the room, 
so ```138-hdmi-audio-27mhz``` => ```138-h...27mhz```, 5chars-dotdot-5chars.

gardners, 2020-09-26:
Sounds reasonable to me.

dansanderson, 2024-07-01:
It looks like this issue can be closed.

For example, a "dirty" build of branch `726-fast-rnf-with-no-disk` generates these version strings:

```
internal: 726-f...-disk,20240701.15,644d278~
freezer:  20240701.15-726fas-644d278
file:     20240701.15-726fas-644d278~
```

Matrix Mode string of formal release build of `release-0.96`: `release-0.96,20240224.00,3c10488`

lydon42, 2024-07-02:
Yes, I fixed version generation for 0.95 and improved it again for 0.96, far as I remember. Thanks for finding this.

closed 2024-07-02 by lydon42