MEGA65/mega65-core #292: scryptos not generating gitstring properly branchname

Ben-401, 2020-09-21:
similar to #291 

The gitstring generated under jenkins (scryptos) shows up as "HEAD (,295c309+DIRTY,20200917.07"

where the branch name is not handled correctly.

This is seen on the hardware when booting (seen on the hypervisor page)
and in the build logs via scryptos.

Ben-401, 2020-09-26:
i can reproduce this behaviour, maybe repo is in detached head...

bluewaysw, 2020-09-26:
Yes, Jenkins uses commit checkout, that seems to end in detached head:

> git checkout -f 7a9f98c97225afc68baa5e047c455f20cf3dda6b # timeout=30

One option would be to let Jenkins pass the branch nach to the make via variable, pass this to getversion and override the git extraction logic.

Ben-401, 2020-09-27:
yes @bluewaysw but we also want to maintain being able to compile locally (without Jenkins).

Id rather ask @frehwagen to see if he can add "git checkout <branchname>" in his build-scripts so that the repo is not detached, as this will make both the Jenkins users and those compiling from home happy.
Yes I understand that for submodules, being in a state of detached head is normal.
Jenkins should happily accept checking out a branchname, opposed to checking out the HEAD commit.

lydon42, 2022-01-04:
Seems to be fixed.

closed 2022-10-02 by lydon42