MEGA65/mega65-core #293: Sprite data pointer can be changed mid-sprite

gardners, 2020-09-22:
Lots of C64 software relies on the feature that the VIC-II latches the sprite pointer value at the start of drawing a sprite.
MEGA65 currently allows changing it at any time. This needs to be made compatible with VIC-II behaviour.

Commando title screen shows doubled sprites for example as a symptom of the problem.

lydon42, 2022-01-07:
This is not fixed. With latest pre release development I still have a broken title screen. COMM and last bit of O are shown above the actual title position again. Also the Text in the footer seems broken.

lydon42, 2022-10-02:
I don't think that there is still a sprite problem in RC1, It looks more like a raster problem? Lower part of the screen should be black showing "PRESS FIRE TO START", but I see the graphics going down all the way instead.

So I close this as solved (COMMANDO title sprites look fine).

closed 2022-10-02 by lydon42