MEGA65/mega65-core #301: Consider adding "CPU I/O port" via the regular I/O space too, and allow switching off legacy 0/1 access

lgblgblgb, 2020-10-11:
One thing inherited from the C64 world carried over on various "C64-compatible" systems (including C65, MEGA65 but even C64DTV) is the "CPU I/O port", ie (physical) addresses of 0/1. Often it's a bit unfortunate that it's there or hard to access if memory is mapped, thus physical 0/1 addresses are not even mapped in from the viewpoint of the CPU, etc etc. But these are only examples.

What I would love to see: having CPU I/O port DATA and DDR registers also mapped through the regular (though in VIC-IV mode only) I/O space with the very same behaviour as the usual 0/1 space. The second part: having another register to be able to switch OFF the legacy 0/1 feature. Yes, that later one can be dangerous if someone switch that off but well ... Also maybe if a feature like this is implemented, probably Hyppo/etc should be modified to always access CPU I/O port DATA and DDR registers via I/O space to avoid confusion (or let mega65-core implement a feature to map back for the time of a hypervisor mode, ie, CPU I/O port is always mapped in hypervisor mode).

I'm not sure how others will want to kill me, because of an unorthodox idea, like this one ;)

lgblgblgb, 2020-10-11:
Btw, just discovered that I mentioned this already in #36 but that's a bit too huge issue combined with a totally different topic, confusing ...

lydon42, 2024-08-17:
As far as I remember MAP will allow you to access 00 and 01, correct? @ki-bo 
Can we close this?

ki-bo, 2024-08-17:
> As far as I remember MAP will allow you to access 00 and 01, correct? @ki-bo 
> 
> Can we close this?

Mapping block 0 with offset 0 easily allows you to access memory addresses 0,1 instead of the CPU port registers. Mapping any other memory to block 0 will also allow to read/write mapped addresses 0 and 1 as expected. This issue can therefore be closed in my opinion.

closed 2024-08-17 by lydon42