MEGA65/mega65-core #303: 1351 mouse "stutters"/"jumps" in GEOS

adtbm, 2020-10-16:
We tested the 1351 in GEOS and compared to the MEGA65 CONFIGURE menu, the mouse "jumps" every couple of pixels a bit.
The Micromys has exactly the same problem.

The AMIGA mouse works fine (but keyboard input is not working when the mouse is plugged in)

I've tried to change to "1351 emulation" in CONFIGURE, but that doesn't help either.

bluewaysw, 2020-10-16:
GEOS currently overrrides MEGA65 CONFIGURE and forces the 1351 emulation mode, to prevent the keyboard being locked after startup. This happens if a 1350 is connected initially (an 1351 emu is off?). The keyboard was released then if the mouse is moved around, not sure if this was still the case in the latest bitstream. Forcing the emu mode for 1351 cause the keyboard to work properly but seems to break if a real 1351 is connected. 

So the emu mode by GEOS is wrong, so we need a fix for the keyboard lock after startup if a 1350 is connected.

gardners, 2020-10-16:
Hello,

There is a way to disable the joystick port from being able to interfere with the keyboard.  Or at least I thought I there was. Trying to find it... Hmm, no, it would seem my memory is playing tricks on me. BUT what you can do in this case, if you suspect that the joystick is causing problems, you can disable the physical joystick ports via writing $00 to $D611 (writing $02 re-enables them). Note that this is a "write only" function, so don't try to set/clear the bits, instead just write the correct value.

I'd recommend that you just leave the mouse emulation as set by configure, to avoid user confusion.  Setting the 1351 emulation to "aggressive" should also greatly ease the problem.  The problem is if the mouse is resting in a position where the quadrature signals are plausible for a joystick. Or alternatively, add an option to your GEOS configure that allows toggling it.

lydon42, 2022-10-02:
#549 closing old issue, keeping the new.

closed 2022-10-02 by lydon42