MEGA65/mega65-core #304: CPU Rework

gardners, 2020-10-23:
The current CPU memory interface is a bit of a mess because of the ISE to Vivado migration.
Basically ISE allowed asynchronous BRAM access via some magic, and the MEGA65 memory interface was built on this. However, Vivado refuses to do the same magic (probably quite sensibly, as it turns out).
The work around was to put some of the magic into the MEGA65's CPU, however this makes it really hard to manage memory accesses, as there is a two-stage process of presenting the data and latching it in two different places. The end result is that the code is a pain to maintain -- although we very much appreciate the work done to work around the Vivado problem.

But the time has come to fix this once and for all.

The best solution is to move the memory interface out into a separate VHDL file, which abstracts all this away, and uses the BRAM only in a synchronous manner.  This will increase memory access latency. So at the same time, we should increase the clock speed to the BRAM and use pipelining to meet timing closure. 

If we add an instruction fetch mode, so that all bytes of an instruction are fetched at the same time, the average latency per instruction should actually drop, as well. Thus we will get more maintainable code, and very likely a reasonable boost to instruction throughput.

Current plan is to drive the BRAM at 8x CPU clock speed, i.e., at 324MHz for 324MB/sec maximum throughput.  By adding an "instruction fetch" flag to the interface, we can make it very quickly fetch whole instructions, and will probably even end up having some little instruction fetch buffer as well, that will mean that the latency per instruction will be encountered only following branches.  If we have a backward-branch instruction fetch buffer, we can probably avoid that latency for tight loops, as well.

This will also make it much easier to interface to the HyperRAM, as we can remove some of the finicky latency-reducing symetry breaks in that interface.

In the end, we will also update the DMA controller to use the fast fetching mechanism, to allow much faster DMAs within the fast RAM.

Finally, all the pipelining will make it possible to have >384KB fast/chip RAM on the MEGA65 R1 and R3 boards that have the A200T FPGA. In theory, we could have about 1MB of fast/chip RAM on those FPGAs.  But with the old design, its not possible to get timing closure to a big BRAM that uses all the little BRAMs, because the physical distances on the chip die are too great.

Current plan is to have 8x 324MHz clocks of latency on data fed to the BRAMs, and the same on the read side. This means two 40MHz ticks on the round trip.

As mentioned, for instruction fetch, we can largely hide this.  For data writes, the write can be scheduled asynchronously.  For data reads, we will now have higher latency than before, by two cycles. However, because the instructions will fetch in 1 cycle typically, the 1+2=3 cycle latency for random reads will still mean we can do an absolute mode read in 4 cycles -- the same as the 6502, and 1 cycle faster than the MEGA65 can do now.  Writes will happen in 1 cycle, however, providing a very significant speed-up.  If we implement the ZP cache, then the ZP indirect modes will be able to do a ZP vector fetch in 3 cycles, the same as now, or possibly even a bit faster, since we can clock the ZP cache at 2x or 4x CPU clock to allow fetching multiple bytes in a single CPU cycle.  This will especially help the 32-bit ZP pointer modes, as their vector fetch time will drop from 5 cycles to probably 2 or 3 cycles.

brouhaha, 2020-10-24:
I wasn't aware that ISE had any magic to make BRAMs look asynchronous, but in both Vivado and ISE I have long done that by clicking the BRAM on the falling clock edge, vs rising edge for all the logic. The usual BRAM setup and hold times still apply, but relative to the falling edge. The static timing analyzer knows how to handle that.

I'm _not_ recommending that you change your plans.

gardners, 2020-10-25:
Hi Eric,
Yes, this all came about because ISE happily synthesised with an async description of the BRAMs in my VHDL, but Vivado refused to do so.  Interesting re clocking on falling edge. I had at some point thought about doing that, but didn't know enough about how it all works to be confident it would be a better solution.

Hopefully I can get the latencies and pipelining under control enough that maximum CPU speed is not worse than at present with my rework. If I can do that, then I won't need to change plans. But if that proves to not be possible, then clocking on falling edge might be a good alternative.

In any case, thanks for the interesting feedback.

Paul.

lydon42, 2024-08-17:
@gardners the cpurework branch seems not to be merged, is this something we can throw away, as it is too old now?

gardners, 2024-08-17:
It can be closed, but don't discard the branch, in case I ever decide to get around to finishing this job @lydon42 .

commit 14377454 (2021-01-26): more timing closure fixes #304
closed 2024-08-17 by gardners