MEGA65/mega65-core #308: $D068-$D06A not being reset by hot registers

FeralChild64, 2020-11-28:
Execute the following code from C64 mode:

```
	; unhide VIC-IV

	lda #$47
	sta $D02F
	lda #$53
	sta $D02F

	; change chargen location

	lda #$01
	sta $D06A ; chargen pointer MSB

	; restore original chargen location by using VIC-II hot register

	lda #$04
	sta $D018
	lda #$14
	sta $D018
```

Result: on the DevKit we get a blank screen (character set is taken from empty C65 RAM - wrong), on XEMU after this piece of code is executed, a normal screen is displayed (expected behaviour). It seems to me that on the DevKit any change to $D068-$D06A result in a state that cannot be reverted to the original VIC-II behaviour any more (wrong).

gardners, 2020-11-28:
This is a feature, not a bug.  $DD00 banks DO get applied, however.

From the VHDL:

```
      -- Apply VIC-II banks
      screen_ram_base(15 downto 14) <= not last_dd00_bits;
      vicii_sprite_pointer_address(15 downto 14) <= not last_dd00_bits;
      character_set_address(15 downto 14) <= not last_dd00_bits;
      
      -- All VIC-II/VIC-III compatibility modes use the first part of the
      -- colour RAM.
      colour_ram_base <= (others => '0');

      -- NOTE: We DONT reset the character set address with a legacy write.
      -- This makes it easier to use alternate character sets, and full-colour
      -- text mode upgrading of old programmes.
```

If you think we SHOULD do it, we can change that. But given that there is very little software that could possibly expect it right now, I'm uninclined to change it.

FeralChild64, 2020-11-28:
Well, actually Open ROMs (the build which is a part of core flashed on the DevKit) expects this, otherwise it's "GO 64" command won't work.

My use case is: I've set the $D068-$D06A to do what I wanted to (in my case it was Open ROMs, but if someone decides to use these registers from within his newly written file browser, he will end up with exactly the same problem), now I want to launch a legacy program. How do I revert all the effects of my writes to $D068-$D06A?

gardners, 2020-11-28:
The easy remedy is to write $001000 back into those registers before launching the legacy programme, or does that leave problems for you?

FeralChild64, 2020-11-29:
Seems to work - but I haven't tested moving charset to another location using VIC-II/CIA in C64 mode yet.

closed 2022-01-01 by lydon42