MEGA65/mega65-core #309: CROM9 bit of VIC control register A not working

FeralChild64, 2020-11-28:
C65 ROM contains 2 character sets - one ant offset $D000 (original C64 character set), and a second one at offset $9000 (the C128 character set). It is possible to switch to C128 character set by setting bit 6 of VIC-III register $D030. This does not work on MEGA65 DevKit. Easiest way to reproduce problem:

- prepare special MEGA65.ROM - wipe character generator at $9000-$9FFF offsets
- most likely after starting the system with working CROM9 bit you should see a blank screen (I suppose the C65 ROM selects $9000 charset for C65 mode); if not, go to C64 mode, and type:

POKE 53295, 71 : POKE 53295, 73 : POKE 53296, 64

now you definitely should see a blank screen (but on MEGA65 DevKit there is no change in the display - wrong).

BTW. It seems that XEMU does not implement this bit either, and Open ROMs currently uses $9000-$9FFF area for other purposes - I'll have to fix this.

gardners, 2020-11-28:
MEGA65 currently doesn't implement this bit. We might in the future, but its not currently a priority.
Even if we do, it will only be on T200-based MEGA65 targets, as it requires an extra 4KB BRAM to have the doubled character ROM.

dansanderson, 2023-08-13:
We recently confirmed that CROM9 is not implemented in the C65 prototype. I'm removing erroneous prose descriptions of CROM9 from the MEGA65 documentation to avoid confusion. We'll have to decide how to interpret historical artifacts for intended behavior, and how to prioritize implementing a feature that is not needed for historical compatibility and is not practical for modern programs. (CROM9 is not used by the BASIC FONT command to switch fonts. Font switching is a MEGA65 addition and works by copying charsets from the ROM to MEGA65 character memory.)

lydon42, 2024-08-17:
@dansanderson should I add (not implemented) to D030.6 description?

gardners, 2024-08-17:
I believe marking it not implemented is the right approach, and then closing it.

closed 2024-08-17 by lydon42