MEGA65/mega65-core #316: Pac-Mania game screen badly distorted

FeralChild64, 2020-12-06:
The Pac-Maina (https://www.lemon64.com/?mainurl=https%3A//www.lemon64.com/reviews/view.php%3Fid%3D615 ) screen is badly distorted, so that the game is unplayable:
- the playfield shakes
- lives left are not displayed properly
- sprite positions are off
- etc.

[image]

gardners, 2020-12-18:
Was the MEGA65 in the correct PAL/NTSC mode?
Was IO mode in C64 or C65 or M65 mode?

FeralChild64, 2020-12-18:
I've checked once again - it's set to PAL, as it should for this game. I used the C65 ROM supplied with the DevKit, in C64 mode (tried both starting machine with C= and switching to C64 mode via GO64) - so I guess IO mode should be C64.

lydon42, 2022-01-03:
Just checked with pre release development core on r3: it's not the same bad, but different. Only the score lines seem to be doubled but the play area scrolls fine.
Also the freezer freezes.
[image]

lydon42, 2024-08-17:
The C64 Mode is not completely compatible that is to be expected, use C64 core instead.

closed 2024-08-17 by lydon42

gardners, 2024-08-17:
Agreed. this is related to the VIC-IV badline handling fu, that is known to have problems. Boulder Dash shows this even worse. C64 core is the workaround for now.