MEGA65/mega65-core #321: DMA texture issue on Nexys 4

headkaze, 2020-12-18:
Exhibited whilst running the MEGAMAZE demo. Screenshots attached show expected vs Nexys 4 rendering.
[image]
[image]

headkaze, 2020-12-19:
Looking at the source we can see the following #define for the location of the textures:
`#define TEXTURE_ADDRESS 0x8000000L`
Which according to iomap.txt refers to
`GS $8000000 - $FEFFFFF Slow Device memory (127MB)`
Which is the expanded memory or PSRAM on the Nexys 4.

Please refer to https://github.com/MEGA65/mega65-core/issues/324

gardners, 2020-12-20:
Yes, so it looks like it is using the expansion RAM for the textures.

lydon42, 2022-10-02:
There is no attic RAM support on the Nexys4. A possible solution is to either support the DRAM on the nexys4ddr / A7, or to allow the use of PMOD hyperram. This is something that could be done, but the focus currently is on the mega65r3 boards.

And already is covered in issue #324, so closing this one.

closed 2022-10-02 by lydon42