MEGA65/mega65-core #322: Erasing flash slots not working on Nexys 4

headkaze, 2020-12-18:
Launching slot 0 works with the latest **138-hdmi-audio-27mhz** branch but erasing a slot or flashing a bitfile to a slot is failing (see attached screenshot).

Also the nexys4.xdc and nexys4ddr.xdc files need ConfigRate changed from 66MHz to 26MHz so flashing to QSPI from Vivado works without CRC errors.

`set_property BITSTREAM.CONFIG.CONFIGRATE 26 [current_design]`

[image]

lydon42, 2022-01-03:
Duplicate #348 ?
nexys4.xdc and nexys4ddr-widget.xdc both have the 26MHz setting. Note that nexys4ddr target is obsolete, please use the -widget variant.

Can you please test with a recent (today) development core?

lydon42, 2022-10-02:
#522

closed 2022-10-02 by lydon42