MEGA65/mega65-core #325: iverilog no longer builds

paulscottrobson, 2020-12-20:
Due to its somewhat antiquated version ; is it necessary to have this iverilog rather than just a standard one ? 

Specifically, bison generates files differently causing parse.hh / fp.hh not to appear in four Makefiles, and the switch to -fno-common default in gcc10 stuffs up the compile.

gardners, 2020-12-20:
To be honest, I can't actually remember why I had it in there. Maybe to better support simulation in GHDL.  I don't see why it can't be updated.

paulscottrobson, 2020-12-22:
Removed it and used iverilog from repo (now a dependency). ImageMagick is also an unmentioned dependency.

Line 749 (roughly) becomes the following - have just removed dependency on cloned iverilog and changed command line to use iverilog in path

```
$(VHDLSRCDIR)/uart_monitor.vhdl.tmp $(VHDLSRCDIR)/uart_monitor.vhdl: Makefile $(VERILOGSRCDIR)/monitor_mem.v
	$(warning =============================================================)
	$(warning ~~~~~~~~~~~~~~~~> Making: $@)
	( cd $(VERILOGSRCDIR) ;iverilog  -tvhdl -o ../../$(VHDLSRCDIR)/uart_monitor.vhdl.tmp monitor_*.v asym_ram_sdp.v 6502_*.v UART_TX_CTRL.v uart_rx.v )
	# Now remove the dummy definitions of UART_TX_CTRL and uart_rx, as we will use the actual VHDL implementations of them.
	cat $(VHDLSRCDIR)/uart_monitor.vhdl.tmp | awk 'BEGIN { echo=1; } {if ($$1=="--"&&$$2=="Generated"&&$$3=="from"&&$$4=="Verilog") { if ($$6=="UART_TX_CTRL"||$$6=="uart_rx") echo=0; else echo=1; } if (echo) print; }' > $(VHDLSRCDIR)/uart_monitor.vhdl

```

closed 2020-12-23 by paulscottrobson