MEGA65/mega65-core #326: Setting CharGen Vertical pos ($d04e) mid frame has no effect

smnjameson, 2020-12-22:
In order to create a static HUD a the bottom of the screen while scrolling a map in the upper section its necessary to allow the changing of $d04e on a raster split. Currently this has no effect:

https://user-images.githubusercontent.com/4640763/102932319-f79dda80-4497-11eb-8291-c12fec5af938.mp4

Video show current behaviour:
Grey border = Raster sets $d04e scrolling all the time
Black Border= Raster sets $d04e to static value but scrolling continues from top area

lydon42, 2022-10-02:
@smnjameson is this fixed in recent cores? Can you provide a simple test program?

archiecrux, 2024-08-24:
[SPLIT.zip](https://github.com/user-attachments/files/16737329/SPLIT.zip)

setting up interrupt handler and therein ...

lda $d079
inc
inc
inc
sta $d04e

... is working