MEGA65/mega65-core #330: DMA transparent bytes appear not to work

smnjameson, 2020-12-28:
I would expect the following code to omit all the copied bytes as they are zero and the transparency is enabled and set to byte 0

`.cpu _45gs02                
#import "./include/m65macros.s"

BasicUpstart65(Entry)

* = $2016
Entry:        
        enable40Mhz()
        enableVIC4Registers()    
        lda #$00
        sta $d702
        sta $d704
        lda #>DMA
        sta $d701
        lda #<DMA
        sta $d705
        rts
Data:
    .fill $400, 0

DMA:
    .byte $0a       //dma type
    .byte $06         //enable transparency
    .byte $86, $00  //Transparent byte
    .byte $00
    .byte $00        //Copy and end    
    .word $0400        //length
    .word Data         //Source
    .byte $00
    .word $0800        //Dest
    .byte $00
    .word $0000        //Modulo`

However the output is as follows:
[image]

gardners, 2020-12-31:
Use $07 to enable transparency. $06 disables it.  Let me know if the documentation is backwards.

smnjameson, 2020-12-31:
That did indeed fix the issue, documentation seems to be incorrect.

lydon42, 2021-12-31:
This is correct in recent documentation.

closed 2021-12-31 by lydon42