MEGA65/mega65-core #331: Virtual F011 is not usable anymore (with monitor_load)

frehwagen, 2020-12-28:
Using the #138 branch (@138/179) up to commit 047d24516123cfe933bfa972fc917cc571e05152 virtual f011 is working when using monitor_load to virtually serve a D81 image. Starting commit ce1efa0b9ee5d15a93989f12a82306e5ab352c4c (@138/180) it stopped working haning a while in the MEGA65 bootup screen, then sometimes later flashes the border and finally stops with broken screen.

frehwagen, 2020-12-31:
Looks like the acutal issue is introduces by this commit:

SHA-1: 9a84385d2dfdeb291fc79f1f2e86fca54ffb2ea4

* stop endless uart retrigger when RX low #295

gardners, 2020-12-31:
ok, that's quite weird if it is the issue.  Maybe try reverting it, and building a bitstream that way, and see if it helps.

frehwagen, 2021-01-01:
OK, looks like reverting the one commit (SHA-1: 9a84385) fixes the communication issue up to 

SHA-1: 5e5f3f2497cf3c023f311304ea28bd05e2ee60a4

* add dummy max10 commit info #317

which is the one short before the buffered UART is introduces (#7), then it fails again in quite the same way. Revering all #7 afterwards makes it working again if it comes to using F011 virtualization.

frehwagen, 2021-01-01:
What needs to change to make the buffered UART working wit monitor_load?

gardners, 2021-01-01:
All those changes have been made to m65.c, m65common.c and mega65_ftp.c in the https://github.com/mega65/mega65-tools repo
Don't use the old monitor_load code any more. This means that VF011 support probably needs to be re-added to that code.  The new code is much easier to use, as there are push_ram() and fetch_ram() routines in there now.

frehwagen, 2021-01-01:
We should remove monitor_load then from the repo if it is not maintained anymore.

gardners, 2021-01-01:
The fix is actually on the mega65-tools repo, and tagged with #331.

gardners, 2021-01-01:
Completely agree, thus started removing them.
Please give the new m65 fixes a go, as I can now load from a virtual D81 at about 8KB/sec at 40MHz in C64 mode, which is not that much slower than with SD card.

gardners, 2021-01-01:
(I've pushed linux binaries to you on discord, if you are still logged in there)

closed 2022-01-03 by lydon42