MEGA65/mega65-core #332: Transparency Issues with 16 color sprites

smnjameson, 2020-12-30:
Reduced a bug I was seeing down to the following reproducable test. As you can see overlapping sprites in 16 color mode (16bit pointers, Color 0 set as transparent) seem to have issues:

https://user-images.githubusercontent.com/4640763/103364107-cf743400-4ab4-11eb-8c7d-3e6b80a93d1c.mp4

Tet prg here:
[SpriteTest.zip](https://github.com/MEGA65/mega65-core/files/5754725/SpriteTest.zip)

gardners, 2020-12-31:
Most curious. Are you able to quantify more exactly what is happening?  It looks like pixels to the left of the overlap are ending up black (or maybe some other colour).  Maybe if you can make a programme that shows the problem statically (maybe hit space to toggle the problem-causing-overlap), then I can dig into it a bit more easily.

smnjameson, 2020-12-31:
It seems to be that in an overlapped area, the sprite underneath takes the left most column of pixels in the underlying area and repeats them for each column in the overlap, in this case showing black for transparency was in the left most column. The following update to the test will move the sprite to the same position when you press fire:

[image]

[SpriteTest.zip](https://github.com/MEGA65/mega65-core/files/5757675/SpriteTest.zip)

gardners, 2021-01-09:
Seems to fix it for me. Please confirm in latest bitstream, then mark STAGED.
Test programme for exercising this issue is in github.com/mega65/mega65-tools /src/tests/test_332.c

gardners, 2021-01-10:
Test written in mega65-tests src/tests/test_332.c

lydon42, 2021-12-31:
Checked with stable bitstream on r3 hardware, all looks good!

closed 2021-12-31 by lydon42

lydon42, 2022-07-31:
Correct: [image]
Wrong: [image]
(wrong example made with mega65r3-20201226.18-138-hd-7534ac3.bit)