MEGA65/mega65-core #337: RRB GOTOX is off by 1

smnjameson, 2021-01-14:
When setting a GOTOX position of X=0 the next RRB layer actually starts drawing at X=1

[image]

Note the grey layer (RRB) does not start at x=0, the first column of pixels in the white layer(base) is completely unobstructed

gardners, 2021-01-16:
I'll try to take a look.

dansanderson, 2023-03-13:
This issue should be closed. It was decided to not correct the issue to preserve compatibility with existing software. See comment from @gardners here: https://github.com/MEGA65/mega65-core/pull/627

(I don't have perms in this repo so I can't close it myself. :) )

lydon42, 2024-08-18:
@dansanderson @ki-bo on 13th dan writes we don't fix it, on 19th there is a commit from ki-bo adding a -1. Did we change our mind and fixed it anyways? I can't remember...

Please clarify by comment and close the issue afterwards!

ki-bo, 2024-08-18:
Indeed, this was discussed again in Steerco and we decided to fix this one along with some other VIC bugs. It is already fixed in development for quite some time and did not seem to provide any issues so far.

lydon42, 2024-08-18:
Yeah, and it's already in 0.96 release.

closed 2024-08-18 by lydon42