MEGA65/mega65-core #339: When enabling 16 sprite pointers, the low byte of the sprite data address is ignored

M3wP, 2021-01-17:
When enabling 16 bit sprite pointers with $D06E, the low byte address of the pointer is not used in the calculation to determine the location of sprite data in memory.

hernandp, 2021-03-12:
Guys, is there any workaround for this ? Thanks.

hernandp, 2021-03-20:
There is any workaround for this? Thanks.

hernandp, 2021-04-26:
@rdpeake @gardners  SPRPTR16 test program.   Press any key to switch between 0x380 with 8-bit pointer table and 0xC000 using SPRPTR16.  

[sprptr16.test.zip](https://github.com/MEGA65/mega65-core/files/6378894/sprptr16.test.zip)

gurcei, 2021-05-04:
I've done some inspection of the test and fiddled with the $D06C-$D06E values on the fly via m65dbg and shared some observations on this discord thread:
- https://discord.com/channels/719326990221574164/783728598313992262/839109186482733056

From what I've seen, it seems as though it is not so much that the LO-byte in $D06C is being ignored, but moreso the first 4 bits of it are being ignored.

gurcei, 2021-05-04:
Further inspection of `viciv.vhdl` within that Discord thread has led me to the following conclusions on the behaviour of sprptr16 location:

So from what I'm seeing here, this 16-byte boundary location for the sprptr home is 'by design'.

So if I were to document it, I'd say:
```
$D06C = 
  - bits 0-3 ignored
  - bits 4-7 = SPRPTRADDR bits 4-7
$D06D =
  - bits 0-7 = SPRPTRADDR bits 8-15
$D06E =
  - bits 0-3 = SPRPTRADDR bits 16-19
  - bits 4-6 = ignored
  - bit 7 = SPRPTR16 enabled flag
```

So, from what I'm seeing, I think SPRPTRADDR can be positioned anywhere within the 1st megabyte (and not within the 1st 8MB of RAM, as currently described in the mega65-book)
These 8 sprite-pointers can then point to sprite-data within the 1st 4MB range (as correctly described in the docs).

closed 2021-09-18 by frehwagen

M3wP, 2021-09-30:
Why was this issue closed?  The issue has NOTHING to do with the location of the sprite pointers but the use of the sprite pointer data at that address.  As far as I can tell, the original issue still persists.

commit ca340514 (2021-01-17): try to fix #339