MEGA65/mega65-core #340: Allow masking of each pixel row to allow for more flexible RRB Y scrolling

gardners, 2021-01-17:
When using the RRB's ability to fine adjust char addresses to scroll in the Y direction, it means you can't easily scroll individual tiles vertically, because you need blank rows of char data above and below, which also basically means you can only vertically scroll tiles that are vertically laid out in a particular way.

To resolve this, I intend to implement a mechanism where the individual pixel rows of each char can be drawn or not drawn. This would allow masking the rows of char data that have come from the previous or next char tile, thus avoiding the need to have blank data there.

geehaf, 2022-08-29:
Hi - a fix test is now available for this (https://github.com/MEGA65/mega65-tools/pull/122) but I'm not sure if we need to update the status/label from "fix test required"? Cheers.

gardners, 2023-01-26:
test_340.c in mega65-tools has been updated and improved and confirms that this issue has been fixed.

geehaf, 2023-01-26:
Hi Paul, Thanks for the update - it looks like you've been busy, extending
the range of tests and fixing the rowmask issue. On Australia day too. Nice
work.
Cheers,

George

lydon42, 2023-01-26:
Running the test with a fixed core gives the same result as with a release core: a fail.

Looks to me like the wrong pixels are read? Or perhaps you don't set video mode correctly (my system defaults to NTSC)?

lydon42, 2023-01-26:
Fixed and beautified test.

closed 2023-05-31 by lydon42