MEGA65/mega65-core #343: leaving CONFIGURE or booting C64 ROM results in blue screen/MEGA65 crash

adtbm, 2021-01-18:
- when i downloaded the C64 kernel, renamed it to MEGA65.ROM and put it on the SD card, the MEGA65 just booted into a completely blue screen, not C64 Kernel

the blue screen looks exactly the same like,

- when leaving CONFIGURE menu, it happens now quite often, that on leaving CONFIGURE instead of doing a reboot, MEGA65 will hang in a complete blue screen.

gardners, 2021-01-19:
For configure at least, the problem is the MEGA65's Hypervisor assuming that it has OpenROMs loaded, and calling a setup routine specific to that.

Commit 6cb6c1545e31a1510b17cf661c865e72ada69b18 might help.

lydon42, 2022-01-03:
How do you test this? Is it still happening?

lydon42, 2022-10-02:
Closing because no answer to request for confirmation. Please reopen if this still happens with recent cores.

closed 2022-10-02 by lydon42