MEGA65/mega65-core #345: Allow Raster-Rewrite Buffer (RRB) to do more layers

gardners, 2021-01-19:
For some reason the RRB doesn't seem to be able to draw as many layers as it should. With ~5K VIC-IV cycles per raster, and that FCM chars should only take about 12 -- 15 cycles each, we should be able to have piles of layers, but its acting as though the FCM chars take about 39 cycles each.  

While I need to investigate the root cause of this, its also possible for us to implement a feature where the RRB is only updated every 2nd raster line if in V200 resolution.  This should double the number of cycles, and not result in any difference in the display, since the same raster is effectively displayed twice anyway, in V200 mode.

gardners, 2021-01-19:
Commit e44bc34e6649d3444b856da4ebfe86d42f8dce6c might also help with allowing more RRB cycles per line.

lydon42, 2024-08-18:
@gardners can we close this?