MEGA65/mega65-core #349: Using RRB_double_line casues artifacts

smnjameson, 2021-01-21:
[image]

Regardless of how much or little is pushed into the RRB buffer, when using double line RRB (with V200 enabled and $d05b set to 0),  some lines at the top of the screen are repeated

gardners, 2021-01-22:
If you enable this in normal text mode at the READY prompt, do you get the same effect?
There is a known problem with this, that I have yet to figure out, but I did find that there was a combination of settings for this that worked without this problem.

gardners, 2021-01-22:
Try $40, $80 or $C0 in the RRB settings register, for example.

smnjameson, 2021-01-22:
When only setting the following VIC registers in VIC4 mode and nothing else:

			lda #$40	
			tsb $d051	
			lda #$00
			sta $d05b

The ready screen duplicates the row of spaces between the color bars and the words "ENGLISH KEYBOARD"

[image]

smnjameson, 2021-01-23:
Ok so it seems this is actually working (witrh the exception of a single duplicated first raster line). To do so you need to enable V400 mode, then setting the YScale to 0 in $d05b will correctly allow twice the time per raster and significantly increase the number of RRB layers that can be processed

closed 2021-02-02 by smnjameson