MEGA65/mega65-core #353: Default audio levels are illogical and cannot be changed?

M3wP, 2021-01-30:
The default audio levels seem to be fixed on reboot.  This is not too much of an issue except for the fact that they are shockingly out of what could be considered reasonable.

I can't seem to attach a screen shot of what they should be.  Here is what it should read from the audio settings of the freezer menu:

0
-2
-3
-6
-6
-79 (although -6 would be reasonable)

0
-6
-6
-2
-3
-79 (although -6 would be reasonable)

gardners, 2021-01-30:
The defaults are all set in src/hyppo/audiomix.asm
There should be routines for setting stereo vs mono default audio settings in there. And using configure menu should change between those on boot, as well.

M3wP, 2021-01-30:
Okay, understood I'll take this on then.  The settings aren't being saved as far as I can tell.

gardners, 2021-01-30:
Audio mixer settings are not saved between reboots.
It would be possible to use some of the config sector to put this information, if you feel like making the configure menu capable of setting the audio levels (or choosing between some presets, as you see fit).

gardners, 2021-01-30:
(If you do, please keep in mind that the MEGAphone has quite a few more outputs and inputs that should be configurable, such as BT in/out volumes, micrphone in volumes, cellular in/out for two modems, and quite possibly some more I have forgotten).
If you do that side, I'll make the freeze menu audio mixer save its changes when modified there, and make HYPPO read the settings in.  For Hyppo efficiency, there should be a number of audio coefficient number + 8-bit coefficient pairs = 2 bytes per value, so that I can just read through them in HYPPO. OR we just use the upper 256 bytes of the config sector to hold the complete set of coefficients might be even easier...

M3wP, 2021-01-30:
I'll just do some diffs on the settings as they are given by the current defaults and what I need to change.

There are quite a few config changes lined up!  I think having panning and volume controls would work much better and maybe only the main outputs in config?

M3wP, 2021-01-31:
Unfortunately, I have found a fundamental flaw in audiomix.asm.  It is not checking the system setting for stereo or mono operation and is simply falling through to setting mono after setting stereo.  This fall through is new to the development branch where a change was made to include the OPL.

I do not know how to test the configuration settings myself.  I need to maintain some compatibility with other devices and setting deltas are better for all of them but there is no way I can see of applying them except as something of a hack.

I will simply allow the program to run as it does now and then patch the final result.  This is totally inelegant but the only solution at the current time.

gardners, 2021-01-31:
syspart.asm is the place where system settings are applied.
That should be setting them correctly. So you will likely see setting stereo, then mono, then whatever is correct.  This should be rationalised to probably setting mono only, then whatever is correct (which will only happen if there is a system partition).

M3wP, 2021-02-03:
I believe I am getting inconsistent results between m65 -a and the audiomix program.  I will need to check by recalculating the values manually.  I am still uncertain if the Headphone L and R slots are the right way around in the listing.  I should be setting them the same, in any case.  

I have found, in exploring the mixer settings, that there are serious issues on the Nexys board.  Testing will need to be done on this hardware.

I don't think it is vital that the settings be saved in configuration at this point so long as the settings are logical.

I will check the values with m65 since this does seem to be consistent with the results of my tests so far.  I have not yet looked at the audiomix program.

M3wP, 2021-02-04:
I have been experimenting with the R2 and Nexys and have found what I think are the best settings.  There are a few issues with the current settings and some bugs in a number of programs.  Now that I have a better grasp of what is required and what is happening, I can look at addressing some of the issues.