MEGA65/mega65-core #354: Place a "READY" signal in cycle count

adtbm, 2021-01-30:
i quote BitShifters and LGBs conversation here:
[https://discord.com/channels/719326990221574164/795760112425369610/805095127642669116]

LGB:
Nice! That is one thing I thought when I mentioned! I mean I guessed there must be some place to help out with simple integer multiply, just I didn't know the actual details. Though with division be aware the timing issue that you need 16 (afaik ...) cycles. Not on Xemu, but for MEGA65 you need that (and here is the point which makes me nervous about things like this: since there is no 'ready' flag, and you write software depends on it ... what happens if the future cpu clock of MEGA65 changes or the timing of the div/mul unit, and it's not 16 cycles anymore but let's say 20? ....)

Bit Shifter:
Yes, I'd prefer a signal too. Hopefully @gardners implements one
But I don't expect, that the cycle count of the division goes up in future versions.

LGB:
You can (I guess!) with the CPU-rework when Paul mentions big performance boosts for the future, maybe. What causes the CPU can be much faster. And thus the division still takes the same time, measured in CPU clocks, it would like it needs more cycles. When in reality the same speed, just the cpu is faster :wink:
if it measured in CPU cycles, I mean, and measured in current ~40MHz CPU clock cycles for the needed time

M3wP, 2021-01-31:
I'm not sure how he'd implement ready except for lots of checks to see if the registers have changed.  As far as I'm aware, the calculations are just being made every time it is possible, quite blindly.

I don't know if you can do it but a potential work-around would be to clear the output registers and then check for a change?  I guess its no good for multiply but for divide it might be useful and that's the one with the delay.  I'm not sure if the output registers can be set, however.

gardners, 2021-02-01:
Simple work-around for now is probably something like this:

```
waitforstablevalue:
LDA LSB-of-result
CMP LSB-of-result
BNE waitforstablevalue
```

I haven't confirmed that this works yet, but it should.
The CPU rework when done is probably unlikely to affect the number of cycles required to wait.  But we will see.

gardners, 2021-02-01:
$D70F bit 7 = divider busy flag
$D70F bit 6 = multiplier busy flag

Commit coming soon.

It would be great for someone to make a test_354.c in mega65-tools repo (copy one of the existing tests to save effort), so that we can put it under regression control (important for when we do finally get the CPU rework done).

lydon42, 2022-01-03:
This is already used by software (and ROM), so it does work. Note that the multiplier busy signal is currently never set by the core, as it does not need to do it (it's that fast).
Test will be added in #506

closed 2022-01-03 by lydon42