MEGA65/mega65-core #355: DEW does not set/clear the N flag

FeralChild64, 2021-01-31:
It seems that DEW does not handle N flag at all (Z flag seems to work), contrary to what is stated in both MEGA65 manual and 65CE02 whitepaper. Example test code (changes screen/border colour in case of unexpected N flag status):

```
	lda #$01
	sta $FC
	lda #$FF
	sta $FB

	; At this point N flag is set, it is expected that DEW should clear it

	dew $FB
	bpl @1
	inc $D020 ; if N is cleared (as expected), this shouldn't execute!
@1:
	lda #$00
	sta $FC
	sta $FB

	; At this point N flag is cleared, it is expected that DEW should set it

	dew $FB
	bmi @2
	inc $D021 ; if N is set (as expected), this shouldn't execute!
@2:
```

FeralChild64, 2021-01-31:
Same problem with INW.

gardners, 2021-02-01:
Staging a commit to fix this.

gardners, 2021-02-04:
Please confirm when fixed, and add the "staged" flag to the issue.
Ideally someone would also write a test case for mega65-tools repo in src/tests/test_355.c :)

FeralChild64, 2021-02-06:
Where can I find the bitstream with the fix?

gardners, 2021-02-07:
Latest experimental builds will be at http://cores.dev.mega65.org/experimental

Paul

FeralChild64, 2021-02-07:
Checked today - it seems to be fixed correctly.

lydon42, 2021-12-20:
Created test_355.prg in mega65-tools to check this.
Checked against master@bbd0391 and development@0f9af7e, all tests pass.

```
Entering unit test mode. Waiting for test results.
2021-12-20T16:36:35.312Z START (Issue#0355, Test #000 - DEW/INW DOES N FLAG)
2021-12-20T16:36:35.312Z  PASS (Issue#0355, Test #001 - #01 DEW $NN -NZ)
2021-12-20T16:36:35.312Z  PASS (Issue#0355, Test #001 - #02 DEW $NN +Z-N)
2021-12-20T16:36:35.313Z  PASS (Issue#0355, Test #001 - #03 DEW $NN +N-Z)
2021-12-20T16:36:35.315Z  PASS (Issue#0355, Test #001 - DEW - 3 TESTS PASSED)
2021-12-20T16:36:35.316Z  PASS (Issue#0355, Test #002 - #01 INW $NN -NZ)
2021-12-20T16:36:35.316Z  PASS (Issue#0355, Test #002 - #02 INW $NN +Z-N)
2021-12-20T16:36:35.317Z  PASS (Issue#0355, Test #002 - #03 INW $NN +N-Z)
2021-12-20T16:36:35.319Z  PASS (Issue#0355, Test #002 - INW - 3 TESTS PASSED)
2021-12-20T16:36:35.322Z  PASS (Issue#0355, Test #000 - TOTAL - 6 TESTS PASSED)
2021-12-20T16:36:35.322Z  DONE (Issue#0355, Test #000 - DEW/INW DOES N FLAG)
>>> Terminating after completion of last test.
```

closed 2021-12-20 by lydon42