MEGA65/mega65-core #356: $D613 and $D614 in iomap.txt and behaviour

lgblgblgb, 2021-01-31:
It seems, `$D613` and `$D614` registers are used by open-rom to scan the keyboard without the usual "joystick interference" and having also the "C65 extra keys" as part of the matrix. However by looking iomap.txt it's confusing, since it says:

    GS $D613 DEBUG:CRTACSCNT Count of cartridge port memory accesses (read only)
    GS $D614 DEBUG:KEYMATRIXPEEK 8-bit segment of combined keyboard matrix (READ)

If possible please make it clear what those registers are for, and also please remove the `DEBUG` tag if this new functionality is planned to be supported on longer term.

Also some questions (my be more a documentation kind of material?):

* $D614 seems to select the row to scan, but not with a bitmask but with values 0-8 (8th being the "C65 row"). What happens if someone writes value greater than 8?
* Why do we need to registers, ie writing `$D614` and then reading `$D613`, it would be enough to have a single register, selecting the row by writing and reading the selected row's state on reading. But anyway, then the next question:
* What effect do you have if you read `$D614` or write `$D613`?

gardners, 2021-02-01:
It should be only $D614: Write the row number and then read the matrix row back.
Values 0 through 9 read the keys, values above that will return $FF, i.e., no key(s) depressed.

gardners, 2021-02-01:
I think no change is required? If happy, please close the issue.

lgblgblgb, 2021-02-01:
Thanks! Hmm, interesting! Since I emulated this by using the scheme I've described using both of `$D613` and `$D614` and it seems how open-rom works, since it actually allowed me to be able to type in open-rom. So at least for me it seems, the current behaviour of MEGA65 uses that scheme if open-rom can be used this way. The other question: what selected row 9 does? 0-7: C64 "normal" matrix, 8: C65 extra keys, but what 9 does?

gardners, 2021-02-01:
So I think you are right. I misread the VHDL.  So I think its write to $D614, read from $D613, right?

lgblgblgb, 2021-02-01:
Yes. And one question was, why it's the case and not happens in the way actually you described as well, looks like more logical and more saving in precious I/O space :) So I liked your description that it "should" work that way, just it seems it's not the case currently, though :) Also, in iomap.txt `$D614` is described quite well, maybe just remove the word "DEBUG", but for `$D613` it's documented with a kind of very different functionality something to do with cartridges? :-O Honestly I was not aware too much about these things, only a forum64 thread made me thinking that I should make it possible to use open-rom with Xemu, so this thread of problems came ...

lydon42, 2022-10-03:
BOUNCE: did one of you derived the correct interpretation? Please update the docs :)

lgblgblgb, 2022-10-03:
I can only tell how I emulate this now in Xemu which seems to work: writing $D614 selects the row to scan, then reading $D613 gives the result of the selected row read. $D614 seems to need the value between 0-8, where 0-7 are the standard C64 positions, and the 8 is the "C65 keys" extra row we can say.

lydon42, 2024-08-17:
Essentially we could swap the DEBUG with UARTMISC and be done.
The latest ROMs use D619 for the keyboard, though.

gardners, 2024-08-17:
Agree, just promote from DEBUG to supported in the documentation.

closed 2024-08-17 by lydon42