MEGA65/mega65-core #362: Improper VIC RASTER ($D012) register behaviour in NTSC mode

FeralChild64, 2021-02-09:
In PAL mode, the RASTER ($D012) register goes from $00 to $FF, afterwards from $00 to $38, and repeats; it's OK, this gives 312 lines.

In NTSC mode, it goes from $00 to $FF, and repeats (next cycle goes from $00 to $FF again) - this is wrong, the next cycle should go from $00 to $05 (if MOS 6567R56A is implemented) or $06 (if MOS 6567R8 is implemented). Real world result: Open ROMs fails to start in NTSC mode, it gets stuck at the beginning of the video system test, see https://github.com/MEGA65/open-roms/blob/master/src/kernal/init/setup_pal_ntsc.s

Code snipped I used to detect the actual behavior:

```
	; Fill-in $C000-$C0FF with 0's, leave 0 in .X

	lda #$00
	ldx #$00
@0:
	sta $C000,x
	inx
	bne @0

	; Wait till raster is $FF
@1:
	lda VIC_RASTER ; $D012
	cmp #$FF
	bne @1

	; Wait till raster is $00
@2:
	lda VIC_RASTER
	bne @1
	sta $C000,x
	inx

	; Store all the rasters different than previous ones
@3:
	cmp VIC_RASTER
	beq @3
	lda VIC_RASTER
	sta $C000,x
	lbeq MONITOR
	inx
	lbeq MONITOR
	bra @3

	; The routine ends with launching the integrated monitor; type 'M C000' inside for the results
```

gardners, 2021-02-09:
Ok. Weird. I'll have to have a look at what could be causing that.

ki-bo, 2023-09-23:
The problem is actually different: in NTSC, the raster counter jumps from 261 to 7, not to 0. It just remains longer at value 261. Therefore, the lbeq MONITOR is not triggered as expected. This is a bug in VIC-IV.