MEGA65/mega65-core #363: exomizer required as a dependancy

Ben-401, 2021-02-10:
seems the 'exomiser' program is used by:
src/tools/utilpacker/utilpacker.c
in 'development' branch, (make bin/nexys4.bit)
but this program is not listed as a dependency, nor automatically installed (as a submodule?).

Where do we get this program from?

build fails at:
```
Makefile:878: =============================================================
Makefile:878: ~~~~~~~~~~~~~~~~> Making: bin/COLOURRAM.BIN
src/tools/utilpacker/utilpacker bin/COLOURRAM.BIN src/utilities/mega65_config.prg src/mega65-fdisk/m65fdisk.prg src/utilities/mega65_keyboardtest.prg
sh: 1: exomizer: not found
Could not read packed utility from exomized.prg
Makefile:878: recipe for target 'bin/COLOURRAM.BIN' failed
make: *** [bin/COLOURRAM.BIN] Error 255
```

Ben-401, 2021-02-10:
my mistake, was building for "nexys4" whereas I should have been building for 'nexys4ddr'

seems 'nexys4' target is not supported. (Can someone confirm?)

'nexys4ddr' does not build, but that's another issue.

Ben-401, 2021-02-10:
I confirm that this issue still exists when building for target 'nexys4ddr".
Refer to issue #364

Is the 'exomiser' program (used in mega65-core) from:
https://github.com/exomiser/Exomiser

gardners, 2021-02-10:
Yes, or from https://github.com/bitshifters/exomizer

gardners, 2021-02-10:
nexys4 and nexys4ddr-widget should both work, but nexys4ddr is deprecated. I should remove it.

gurcei, 2021-06-19:
Adding a cross-link to a similar ticket #347

lydon42, 2022-01-03:
#514

closed 2022-01-03 by lydon42