MEGA65/mega65-core #365: VIC in bitplane mode expects sprite data in bank 4

Edilbert, 2021-02-17:
VIC is programmed by the graphics routines of BASIC (92xxxx ROMs) to use bank 4 & 5 for bitplane memory.
If the graphics routines are not called and sprites are used with the text screen, VIC reads them from $600 on bank 0.
If however a graphics screen was opened, VIC looks for sprite data on $600 on bank 4 (the even bitplane bank).
This is bad, because then the lower 8K segment cannot be used for bitplanes, because of interference with sprite data.
Also after closing the graphics screen and returning to text mode, the VIC still wants the sprite data in bank 4.

Suggestion:
The sprite data bank should be fixed to bank 0 only.

gardners, 2021-02-28:
Please provide an example programme that tickles this.
Is it the sprite pointer list that is moved to bank 4, or the sprite data itself?

gardners, 2021-02-28:
I believe this has fixed it. Awaiting test case from @Edilbert to confirm definitively.

Edilbert, 2022-01-08:
Could be solved with an update of the ROM routine SCREEN SET.
Sprite data stay in bank 0, the table of sprite pointers is updated in SCREEN SET.

closed 2022-01-08 by Edilbert