MEGA65/mega65-core #368: Inconsistency in $d018 <-> CHARPTR relation with HOTREG enabled

steph72, 2021-03-04:
After a RESET, when HOTREG is enabled, the contents of CHARPTR are $01000 for uppercase/graphics font ($d018 is $24) and $01800 for lowercase. ($d018 is $26).

Setting CHARPTR to $4f000 and then toggling $d018 again, CHARPTR gets reset to $41000. Which means, the "bank" portion of CHARPTR stays stuck in bank 4. 

Expected behaviour: With HOTREG enabled, toggling $d018 should reset *all* of CHARPTR. 
A BASIC10 program demonstrating the issue is attached (needs ROM 920139 or later to work).

[charptr.prg.zip](https://github.com/MEGA65/mega65-core/files/6084131/charptr.prg.zip)

lydon42, 2021-12-30:
Can confirm this with latest devel on hardware.
Writing to D018 will only set bits 13-0 of character_set_address in viciv.vhdl, bits 23-14 are untouched.

lydon42, 2021-12-30:
Question is: does xemu do it right? Because xemu seems to always zero out everything, regardless of the hotreg setting.
I implemented a patch that does this:
[image]
Looks good?

lydon42, 2022-01-01:
This might not need a fix... see #308

lydon42, 2022-01-01:
This is not a bug, this is a feature.

closed 2022-01-01 by lydon42