MEGA65/mega65-core #370: compare slot 1 against slot 0 bitsstream version.

adtbm, 2021-03-16:
This is a Feature request.

i would suggest to compare the bitstream version in slot 1 against slot 0

if slot 1 MEGA65 Bitstream version is older or not a MEGA65 bitstream, boot from slot 0
slot 0 can/should only be a MEGA65 bitstream (it is a MEGA65 ;-) )

or a different approach:
maybe would it be possible to select the autoboot slot in the flashmenu ?
i.e. i have:
- in slot 0 the nominal MEGA65 bitstream
- in slot 1 Gameboy
- in slot 2 ZX uno
- in slot 3 and alternate MEGA65 bitstream
Now i want to make a test session with the ZX uno, so that i can switch the autoboot slot to 2 ?

So whenever i power on the MEGA65 it will boot from the selected slot, if a valid bitstream is existent, otherwise it will fallback to slot 0.
 Untill i change/deselect the autoboot slot again.

gardners, 2021-03-18:
A timestamp field in the COR files would allow for an efficient "newness" test.

In terms of setting the default slot examined, this is also possible, by storing this elsewhere in the flash (this means we have to erase the 64KB page that holds this, and reserve 64KB for it). A complication is if the newness test is also used, as if you want to auto-start a bitstream that is older than the latest factory bitstream will mistakenly trigger the "do you really want to load this older bitstream?" behaviour. I'd welcome ideas on how to best manage that.

adtbm, 2021-03-18:
I would say, if you can set the default slot (0 - 3) then there is no need anymore to compare the bitstream version, because then it could be intentional that you want to run an old bitstream (maybe for testing, from i.e. slot 2). 
as long as you can set the default bootslot between 0 and 3 this should be fine and as long as slot 0 is always a MEGA65 bitstream and in case of problems it always falls back to slot 0.
So if you set the default slot to 0, the MEGA65 will always boot from slot 0 regardless of what cores are flashed in the other slots, and so on.
This would also prevent the problem we had last time, when we flashed with vivado a new bitstream to the MEGA65, while the previous user updated the MEGA65 only on slot 1, unfortunately we have not been able to erase the core in slot 1 (with CTRL & 1) so the only way to bypass slot 1 autoboot was by doing a full erase with vivado.

just to ensure again: if the selected default slot is empty or not a valid core, it falls back to slot 0 which is always a MEGA65 bitstream.

gardners, 2021-03-19:
If a slot is invalid, unfortunately it will (by the nature of the FPGA
hardware) keep searching through the flash for the next valid slot.

Also, we need to deal with the case where you had the problem with updating
the factory core. I suggest that we should remember the default chosen
core, and the version of the factory core at the time. If the factory core
is updated to a newer core than the default one selected, then it should
ask for confirmation on next boot, and if accepted, we update the stored
version of the factory core in this, so that we don't get asked again.

lydon42, 2022-01-03:
There is now the option to choose between slot 1+2 via dip switch 4: #443

lydon42, 2023-09-24:
Default slot selection is in 683-cartflash branch.

closed 2023-09-24 by lydon42