MEGA65/mega65-core #377: src/tools/fpgajtag/fpga.h:24:33: warning: large integer implicitly truncated to unsigned type

HarlodHolt, 2021-05-01:
When running the following  make src/tools/monitor_load
```
gcc -Wfatal-errors -Wall -g -std=gnu99 -g -Wall -I/usr/include/libusb-1.0 -I/opt/local/include/libusb-1.0 -I/usr/local//Cellar/libusb/1.0.18/include/libusb-1.0/ -o src/tools/monitor_load src/tools/monitor_load.c src/tools/fpgajtag/fpgajtag.c src/tools/fpgajtag/util.c src/tools/fpgajtag/process.c -lusb-1.0 -lz -lpthread
```
This stalls on:
```
In file included from src/tools/fpgajtag/fpgajtag.c:45:0:
src/tools/fpgajtag/boundary_scan.c: At top level:
src/tools/fpgajtag/fpga.h:24:33: warning: large integer implicitly truncated to unsigned type [-Woverflow]
 #define DITEM(...) ((uint8_t[]){sizeof((uint8_t[]){ __VA_ARGS__ }), __VA_ARGS__})
                                 ^
src/tools/fpgajtag/boundary_scan.c:142:38: note: in expansion of macro ‘DITEM’
 static uint8_t boundary_ppattern[] = DITEM(BOUNDARY_PPAT);
```
I am not sure how to fix this but my understanding of warning: large integer implicitly truncated to unsigned type is that this is signed and has a MSB  set and truncates one of the bits off.

gardners, 2021-05-01:
It might work to add a type cast to int to the sizeof() call, e.g.:

```
((int)sizeof(uint8_t[]))
```

HarlodHolt, 2021-05-08:
Okay so this seems to have fixed that issue 
`#define DITEM(...) ((uint8_t[]){(int)sizeof((uint8_t[]){ __VA_ARGS__ }), __VA_ARGS__}
`

HarlodHolt, 2021-05-08:
Now I need to work through all of these
`Makefile:780: =============================================================
Makefile:780: ~~~~~~~~~~~~~~~~> Making: src/tools/monitor_load
src/tools/monitor_load.c:72:64: error: unknown type name ‘uint32_t’; did you mean ‘u_int32_t’?
 void init_fpgajtag(const char *serialno, const char *filename, uint32_t file_idcode);
                                                                ^~~~~~~~
                                                                u_int32_t
src/tools/monitor_load.c:152:1: error: unknown type name ‘uint32_t’; did you mean ‘u_int32_t’?
 uint32_t zap_addr;
 ^~~~~~~~
 u_int32_t
src/tools/monitor_load.c: In function ‘process_line’:
src/tools/monitor_load.c:597:4: warning: implicit declaration of function ‘do_exit’; did you mean ‘_exit’? [-Wimplicit-function-declaration]
    do_exit(0);
    ^~~~~~~
    _exit
src/tools/monitor_load.c:854:32: error: ‘screen_address’ undeclared (first use in this function)
       if ((addr==0xffd3058)&&(!screen_address)) {
                                ^~~~~~~~~~~~~~
src/tools/monitor_load.c:854:32: note: each undeclared identifier is reported only once for each function it appears in
src/tools/monitor_load.c:860:2: error: ‘next_screen_address’ undeclared (first use in this function); did you mean ‘screen_address’?
  next_screen_address=screen_address;
  ^~~~~~~~~~~~~~~~~~~
  screen_address
src/tools/monitor_load.c:861:9: error: ‘screen_width’ undeclared (first use in this function); did you mean ‘screen_address’?
         screen_width=b[6];
         ^~~~~~~~~~~~
         screen_address
src/tools/monitor_load.c:862:2: error: ‘screen_line_step’ undeclared (first use in this function); did you mean ‘screen_width’?
  screen_line_step=b[0]+(b[1]<<8);
  ^~~~~~~~~~~~~~~~
  screen_width
src/tools/monitor_load.c:863:2: error: ‘screen_rows_remaining’ undeclared (first use in this function)
  screen_rows_remaining=25;
  ^~~~~~~~~~~~~~~~~~~~~
src/tools/monitor_load.c:864:2: error: ‘screen_line_offset’ undeclared (first use in this function); did you mean ‘screen_line_step’?
  screen_line_offset=0;
  ^~~~~~~~~~~~~~~~~~
  screen_line_step
src/tools/monitor_load.c:871:24: error: ‘screen_line_buffer’ undeclared (first use in this function); did you mean ‘screen_line_offset’?
  for(int i=0;i<16;i++) screen_line_buffer[screen_line_offset+i]=b[i];
                        ^~~~~~~~~~~~~~~~~~
                        screen_line_offset
src/tools/monitor_load.c:879:6: warning: implicit declaration of function ‘print_screencode’; did you mean ‘print_spaces’? [-Wimplicit-function-declaration]
      print_screencode(screen_line_buffer[i]);
      ^~~~~~~~~~~~~~~~
      print_spaces
src/tools/monitor_load.c: In function ‘main’:
src/tools/monitor_load.c:1529:3: warning: implicit declaration of function ‘init_fpgajtag’ [-Wimplicit-function-declaration]
   init_fpgajtag(serial_port, bitstream, 0xffffffff);
   ^~~~~~~~~~~~~
src/tools/monitor_load.c: At top level:
src/tools/monitor_load.c:1643:6: warning: conflicting types for ‘do_exit’
 void do_exit(int retval) {
      ^~~~~~~
src/tools/monitor_load.c:597:4: note: previous implicit declaration of ‘do_exit’ was here
    do_exit(0);
    ^~~~~~~
src/tools/monitor_load.c: In function ‘run_boundary_scan’:
src/tools/monitor_load.c:1459:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
src/tools/fpgajtag/fpgajtag.c:73:73: error: expected ‘)’ before ‘;’ token
             CONFIG_TYPE1(CONFIG_OP_READ, CONFIG_REG_STAT, 1), SINT32(0));
                                                                         ^
src/tools/fpgajtag/fpgajtag.c:72:17: warning: ‘rstatus’ defined but not used [-Wunused-variable]
 static uint8_t *rstatus = DITEM(CONFIG_DUMMY, CONFIG_SYNC, CONFIG_TYPE2(0),
                 ^~~~~~~
src/tools/fpgajtag/fpgajtag.c:71:50: warning: ‘idcode_len’ defined but not used [-Wunused-variable]
 static uint32_t idcode_array[IDCODE_ARRAY_SIZE], idcode_len[IDCODE_ARRAY_SIZE];
[... 8 lines cut ...]
 static int debug, verbose, skip_idcode, match_any_idcode, trailing_len, first_time_idcode_read = 1, dc2trail, interface;
                                                                                                               ^~~~~~~~~
src/tools/fpgajtag/fpgajtag.c:69:101: warning: ‘dc2trail’ defined but not used [-Wunused-variable]
 static int debug, verbose, skip_idcode, match_any_idcode, trailing_len, first_time_idcode_read = 1, dc2trail, interface;
                                                                                                     ^~~~~~~~
src/tools/fpgajtag/fpgajtag.c:69:73: warning: ‘first_time_idcode_read’ defined but not used [-Wunused-variable]
 static int debug, verbose, skip_idcode, match_any_idcode, trailing_len, first_time_idcode_read = 1, dc2trail, interface;
                                                                         ^~~~~~~~~~~~~~~~~~~~~~
src/tools/fpgajtag/fpgajtag.c:69:59: warning: ‘trailing_len’ defined but not used [-Wunused-variable]
 static int debug, verbose, skip_idcode, match_any_idcode, trailing_len, first_time_idcode_read = 1, dc2trail, interface;
                                                           ^~~~~~~~~~~~
src/tools/fpgajtag/fpgajtag.c:69:41: warning: ‘match_any_idcode’ defined but not used [-Wunused-variable]
 static int debug, verbose, skip_idcode, match_any_idcode, trailing_len, first_time_idcode_read = 1, dc2trail, interface;
                                         ^~~~~~~~~~~~~~~~
[... cut ...]
src/tools/fpgajtag/fpgajtag.c:69:19: warning: ‘verbose’ defined but not used [-Wunused-variable]
 static int debug, verbose, skip_idcode, match_any_idcode, trailing_len, first_time_idcode_read = 1, dc2trail, interface;
                   ^~~~~~~
src/tools/fpgajtag/fpgajtag.c:69:12: warning: ‘debug’ defined but not used [-Wunused-variable]
 static int debug, verbose, skip_idcode, match_any_idcode, trailing_len, first_time_idcode_read = 1, dc2trail, interface;
            ^~~~~
make: *** [src/tools/monitor_load] Error 1
`

gardners, 2021-05-08:
Looks like missing ctypes.h, perhaps

HarlodHolt, 2021-05-08:
I wasn't missing my ctypes.h but I did add `#include <stdint.h>` to  src/tools/monitor_load.c: and that fixed the  unknown type name ‘uint32_t

I do not understand C well enough to understand why this is not compiling:
`compilation terminated due to -Wfatal-errors.
src/tools/fpgajtag/fpgajtag.c:72:137: error: expected ‘)’ before ‘;’ token
 static uint8_t *rstatus = DITEM(CONFIG_DUMMY, CONFIG_SYNC, CONFIG_TYPE2(0), CONFIG_TYPE1(CONFIG_OP_READ, CONFIG_REG_STAT, 1), SINT32(0));
`

I believe it may be with how this is defined:
`#define DITEM(...) ((uint8_t[]){(int)sizeof((uint8_t[]){ __VA_ARGS__ }), __VA_ARGS__}`

gardners, 2021-05-09:
Well done. Yes, the type-cast on DITEM might well be problematic.

lydon42, 2021-12-21:
monitor_load is no part of core anymore, closing.

closed 2021-12-21 by lydon42