MEGA65/mega65-core #379: HyperRAM read failure

steph72, 2021-05-02:
On the R3 board, when writing to one address and reading from a different address, sometimes the last written value is returned instead of the actual value stored at the read address when both addresses are located in HyperRAM.

After writing a different value to the first address, the value at the second address switches to the correct value.

Attached is a BASIC65 program that demonstrates the issue.

[hyperbug.zip](https://github.com/MEGA65/mega65-core/files/6411534/hyperbug.zip)

Edilbert, 2021-05-02:
I can confirm this issue.
The program runs with no errors on XEMU, but shows read failures on the MEGA65 dev kit.
When I inspect the attic RAM with the BS monitor after running the program, its shows correct values.
So it seems, that under certain conditions the read access does not return the value from the correct memory page.

lydon42, 2021-12-30:
BASIC test is a little bit faulty, NEXT R is missing between 200 and 210.
Runs without fault. Fix already merged with master.

closed 2021-12-30 by lydon42

lydon42, 2022-08-13:
Unittest for this is test_585