MEGA65/mega65-core #380: Not able to enter Flash menu

adtbm, 2021-05-03:
i have a strange issue with the last two bitstreams for R3 (the two in the posts above my report), whenenver i try to go to the FLASH menu, instead of booting into the flashmenu (when holding NO SCROLL while powering up), i get instead in the Hypervisor/CONFIGURE menu.
When i boot up the MEGA65 in this condition it gives me on the screen the number "4" when i press "5".
(This happens only with a newly formatted SDcard)
completely discharging the MEGA65 and inserting the "original" SDcard it then works again.

adtbm, 2021-05-03:
Pauls reply:
It sounds like the ASCII keyboard scanner is having marginal timing for some reason.

gardners, 2021-05-03:
Just tried with latest bitstream build here, and it works. Commit id: a1f4135d53c157a9bfdcf249e51ed11b896760d6
Not sure if there is something odd going on. Poke me on discord for latest bitstream I generated here.

adtbm, 2021-05-03:
it does looks like a keyboard timing issue.
https://youtu.be/QexOtJ_k_0A

adtbm, 2021-05-03:
keyboard tester.
- pressing and holding "5"-key
[image]

adtbm, 2021-05-04:
defective cable was the root cause.

closed 2021-05-04 by adtbm

gardners, 2021-05-04:
Loose cable was probably causing glitched clock pulses and/or reflections that were causing the first few keys in the data vector to be corrupted on each transfer. The keyboard protocol runs at several MHz, with an effective scan update frequency of about 10KHz -- 100KHz depending on temperature and phase of moon in keyboard CPLD.