MEGA65/mega65-core #388: ghdl/ghdl_mcode:error: importation has failed due to compilation error

UdayZorawarSingh, 2021-05-18:
When I try to do ```make simulate```  in the development branch, I receive following error:

src/vhdl/colourram.vhdl:28:35:error: primary expression expected

[image]

When I had a look at the line number and column number, it pointed towards the closing parenthesis as shown in the image below

[image]

When I looked at the template file for **src/vhdl/colourram.vhdl** which is present at location **src/tools/makerom/colourram_template.vhdl**, I found that value of ROMDATA is not present in the generated file

[image]

How could this be resolved? I am using Ubuntu 18.04.5.

waym0002, 2021-05-19:
I can get it to compile by removing the assignment completely, i.e. putting `ram : ram_t;` there instead. This seems to work as later on in the vhdl the ram variable gets assigned to so it doesn't have to be assigned at first seemingly.

gardners, 2021-05-20:
The problem here is that exomizer has not been installed, and so when it
tries to make the COLOURRAM.BIN file to then autogenerate colourram.vhdl,
it all fails.  It would be great to have exomizer also setup as a git
sub-module and a rule that checks for its existence, so that this problem
can be totally avoided.

Paul.

UdayZorawarSingh, 2021-05-21:
Thank you Paul.
It's working now.

closed 2021-05-21 by UdayZorawarSingh