MEGA65/mega65-core #394: make error  - settings64.sh not found in vivado_wrapper

Alaa-alsh0581, 2021-06-02:
I invoked make in _master_ branch (01.06.2021) but received an exception in **vivado_wrapper** that _settings64.sh_ not found. I have installed Xilings Vivado v2020.2 and _settings64.sh_ is not in _/opt/Xilinx/Vivado_, I found _settings54.sh_ inside _/opt/Xilinx/Vivado/Vivado/2020.2/settings64.sh_
I know this is not a huge issue but I believe it should be explicitly mentioned in the build documentation.

gurcei, 2021-06-21:
Taking a look at the script in question:

https://github.com/MEGA65/mega65-core/blob/development/vivado_wrapper

What it attempts to do seems reasonable to me, as from my past vivado installations, it would always install to a path such as this:

`/opt/Xilinx/Vivado/2018.2`

So during the install, I stuck with the default path of "/opt/Xilinx/":

[image]

Given that, it settled for a default full-path of `/opt/Xilinx/Vivado/2018.2`:

[image]

So I'm wondering if during installation, you modified the default path to be "/opt/Xilinx/Vivado", and hence why your full-path had two "Vivado" in it, and why the vivado_wrapper script failed to work.

As for how it could be resolved, perhaps you could move the installed items back to the default locations and the problem will go away?

You mentioned the need to explicitly document such things, and I agree that explicit steps for installing xilinx were lacking for quite a while in the project, but I do believe somebody was underway with adding such steps into the mega65 user guide, so hopefully things will get easier in this area in future.

gurcei, 2021-07-01:
I've merged in Amok's installing-vivado's steps into the mega65-book.pdf, so feel like we've made an effort to document the desired steps to avoid such trip-ups in future. Will close the ticket.

closed 2021-07-01 by gurcei