MEGA65/mega65-core #397: Xilinx Vivado installation error

Jord0094, 2021-06-08:
An error I ran into multiple times wasn't for the mega65 however the Xilinx Vivado installation. when installing the vivado i would get to nearly installing and it would freeze on generating installed device list. I would leave this for hours and it wouldn't change. there's very little to fixing this online and a mention of this maybe in the build.md or a fix solution to these kinds of problems would be very useful.

gurcei, 2021-06-21:
Don't have any immediate ideas on what the problem is, but would you be able to capture a few more specific details of your situation on this card.

Which OS are you installing on?
Which version of Vivado are you installing?
Could you share a screenshot of the exact point in the install wizard where it fails?

gurcei, 2021-07-01:
Haven't heard back on this topic for a while, so will have to close the ticket

closed 2021-07-01 by gurcei