MEGA65/mega65-core #398: unable to "make simulate"

ENGR3891-vasu0007, 2021-06-12:
src/vhdl/vfpga/vfpga_wrapper_8bit.vhdl:114:22:error: unit "arch8x6w16n4i10k4fci4fco8pfi8pfo8iopb2_wrapper_8bit" not found in library "work"
src/vhdl/vfpga/vfpga_wrapper_8bit.vhdl:133:25:error: unit "vfpga_clock_controller_pausable" not found in library "work"

i tried to add these in the Makefile but that's throwing an error.

clar0867, 2021-06-17:
Also getting the same error as you with this.
Looking at https://forums.xilinx.com/t5/Design-and-Debug-Techniques-Blog/Using-the-quot-work-quot-library-in-VHDL/ba-p/942871 appears that using work as the library is not advised by Xilinx.
I've tried as you have to manually add within the Makefile which doesn't resolve the issue, but additionally attempted to copy the vfpga_wrapper_8bit file to src/vhdl as this would be assumed as the "work" location, with that not being successful either.

gurcei, 2021-06-21:
As I noticed some similarities with ticket #409, can I confirm if you're working from the 'development' branch? (as that's our most active branch at present)

gurcei, 2021-07-01:
Ok, as we haven't heard back anything on this ticket, and there are users that have successfully used "make simulate" on 'development' branch recently, I'll close this ticket.

closed 2021-07-01 by gurcei