MEGA65/mega65-core #399: Frogs doesn't run on NTSC

matthewpomar, 2021-06-13:
Sorry if this isn't the right project to post this in. I'm reporting that the included "Frogs" program just loads to a black screen when the DevKit is configured for NTSC. I was hoping the r3 core would fix this issue, but alas, the problem persists.

If there's anything I can provide to support this report, please let me know.

Here's a bit of background:

https://www.forum64.de/index.php?thread/110035-how-to-fix-issue-with-sprites-not-properly-aligned-on-devkit/&postID=1609244#post1609244

lydon42, 2021-12-20:
The game in question: https://frogs.drwuro.com/

deftmega, 2022-01-03:
please report to Dr. Wuro, the maker of the game.

lydon42, 2022-01-04:
Retested on r3 with pre release bitstream and rom: Runs in PAL mode, does not run in NTSC.
The only thing that looks incorrect is the radio at the top. Sound is ok, frogs and positions are ok, gameplay works. Can't say how the radio is made, and won't dive into the game code.
The MEGA65 will not have a perfect C64 emulation. For this a future C64 core is the best solution.

closed 2022-01-04 by lydon42

lydon42, 2022-01-05:
In #358 there is a comment about a [NTSC detect method](https://github.com/MEGA65/mega65-core/issues/358#issuecomment-775512425) that causes a hang. Perhaps this can help.