MEGA65/mega65-core #402: DMA Line Fetch - incorrect results

smnjameson, 2021-06-15:
[image]

The above image is created by horizontally scanning a source texture and horizontally drawing it line by line to the screen. The screen is set up with vertical char strips and the source texture is a regular image scanned left to right and top to bottom. In the above case both slopes are X positive with slope values of $00:$00 and it draws correctly

If we take the source slope and change it to $FF:FF it should read the source texture with diagonal lines at a 45 degree downward angle to the original. Drawing these lines out to the screen should have the effect of skewing the resulting screen image 45 degrees upwards. However this is the output:

[image]

Likewise we can rotate a scan line in the source image around a point like a clock hand and draw that texture source data at the same rotated angles on the screen. Essentially copying data in a circle from source to destination. However the result is such:

https://media.discordapp.net/attachments/782757495180361778/854378086216499260/unknown.png

With only the left hand quadrant somewhat displaying the correct data.

smnjameson, 2021-06-15:
Here is a good image showing the problem:

[image]

The same image is scanned and rendered to the screen using lines three times with gaps between each scan line to help visualise the scan line directions. The arrows above each image show the direction of the scan lines and the lines drawn are of the same direction and angle as their corresponding scan line. The orange arrows indicate the direction of the start point of each line over time, with the start point remaining stationary on the third radial example.

You can see that horizontally scanning works perfectly, this is the same as the source texture image.

Scanning vertically does something strange, while the starting point is correctly moving from left to right, it appears like each source line is still taken _**horizontally**_ and so we see the red line at the top of the source image drawn vertically and getting smaller as the source start point moves across it. 

Therefore when scanning at a range of angles as shown in the third image, the results are completely incorrect

smnjameson, 2021-06-15:
[image]
 OK Ignore this!!!

the key is to make sure the texture data is stored like the screen data, in vertical strips of 8x8 chars
obvious really, I feel so stupid

Closing this "bug" as its working as intended

closed 2021-06-15 by smnjameson

gardners, 2021-06-15:
You can actually structure the source data in a variety of different ways, provided you configure the DMA controller to know what you are doing :)