MEGA65/mega65-core #405: make: ***No rule to make target 'ghdl/ghdl_mcode', needed by 'simulate'. stop

Drashti287, 2021-06-17:
[image]

**STEPS:**
1. goto ghdl directory
2. run `./configure`
3. run `make`
4. try again with `make simulate`

gurcei, 2021-06-21:
I noticed there was a recent commit to 'development' branch relating to adding such a target:

https://github.com/MEGA65/mega65-core/commit/73a2cab629aabd9188f83d943ecdeb1ed30ee145

So if you're not presently working from the 'development' branch, I suggest switching to it and trying again from there and see how you go.

gurcei, 2021-07-01:
Ok, as we haven't heard back anything on this ticket, and there are users that have successfully used "make simulate" on 'development' branch recently, I'll close this ticket.

closed 2021-07-01 by gurcei