MEGA65/mega65-core #406: Make simulate Crashing due to Germam Characters

clar0871, 2021-06-19:
I noticed that on rare occasions the make simulate command results in a crash when making the following file.

src/vhdl/matrix_to_ascii.vhdl

After doing some digging, it turns out the German characters used within it are not recognized by the compile resulting in the crash.

Since the characters appear in comments they can easily be removed fixing the issue.
It should be noted that not all compilers result in this issue and when testing with others this crash did not happen.

gardners, 2021-06-19:
Would you like to pose a pull request to fix it?
If the german characters are ü, ä or ö, these can be replaced with ue, ae or oe, respectively.
If it is ß, it can be replaced with ss.
If it's Pippi Longstocking, then she is a swedish character, not a german one.

gurcei, 2021-06-21:
It was later found that this matter has been resolved in the 'development' branch already, so we can close this ticket now.

closed 2021-06-21 by gurcei