MEGA65/mega65-core #408: Is it possible to bypass Xilinx Vivado ?

shinnapattsu, 2021-06-19:
Hi, the step to "make bin/mega65r3.bit" required me to install Xilinx Vivado.
However, the problem is that I do not have enough free disk space to install Xilinx Vivado.
The image below requires about 65 GB but, I have to increase the free disk space up to 100GB to install this program. 

[image]

For this reason, Is it possible to bypass Xilinx Vivado or make this program think i already installed vivado but actually not ?

Thank you very much.

HarlodHolt, 2021-06-19:
Hi shinnapattsu,

You will still need Xilinx Vivado for the  "make bin/mega65r3.bit" to work but to solve your problem you can try a few things.

1. Download and install an older version of  Xilinx Vivado, 2018 webpack, I am running this and it is working.
https://www.xilinx.com/member/forms/download/xef-vivado.html?filename=Xilinx_Vivado_SDK_Web_2018.3_1207_2324_Lin64.bin
2. Launch your installer with sudo or as root, in order to access opt write permission you need to be root. This will allow you to write into /opt folder and get rid of that red text.

Hope that helps,

Yuri

BAKE0391

shinnapattsu, 2021-06-19:
Hi, HarlodHolt,

Actually, I have already installed Xilinx Vivado  2020.2 on my desktop computer.
However, it is inconvenient when traveling around, so I decided to install Xilinx Vivado on my laptop instead.
But, I faced this issue and really curious if there is a way to bypass Xilinx Vivado.
Anyway, I will install an older version of Xilinx Vivado that you have recommended and try for the "make bin/mega65r3.bit" again.

Thank you for your recommendation.

gardners, 2021-06-19:
Xilinx Vivado is the compiler that builds the bit files. You can't compile the bit file without the bit file compiler. Ergo, Vivado is required, however inconvenient this might be to us all.

closed 2021-06-19 by gardners